isofit/isofit #305: Isofit Shape Mismatch & Keyerror '2'

brunohlopez, 2022-11-03:
I am trying to run Isofit on an aviris classic Flightline f171207t01p00r13rdn_e but ran into two problems. The first was that this data had a file called ort_igm, which would signify that the iGM was orthorectified but upon opening it the IGM was actually un-orthorectified. Here is a screenshot of the hdr files for the igm and the img. These should theoretically have the same lines and samples. 
[image]
[image]

 So if you try to run Isofit using the iGM ,obs_ort, and IMG you will run into a shape mismatch error. I'm not sure if this was a data-preprocessing error, but no plain 'igm' file exists in this Flightline either. I also discovered the same thing in multiple similar flight lines. You can fix the issue by mapping the igm to the GLT but since I noticed this across multiple flies, not sure if any other ones were affected 

The second is a 'keyerror' 2 issue that I ran into after I orthorectified the igm. 
[image]

Upon investigation, it seems that the metadata had these values at data type 2, which would be a 2 = Integer: 16-bit signed integer. According to the error the data needs to be a float 32. So I change the data type to a np.float 32 but then I run into a Nonetype data error, which makes me think the image is corrupted after doing this.

[image]

pgbrodrick, 2022-12-05:
Hi @brunohlopez.  Thanks for posting!  As you identified, isofit expects the igm, obs, and radiance files to all have the same dimension and to be aligned - they can be orthorectified or not, but all need to follow the same convention.  I believe the shape mismatch error that you found is one we implemented in isofit to safeguard and report out on this, so I'll chalk this one up to a preprocessing requirement.

Datatypes are handled correctly within isofit so far as I can see.  The failure above is a bit odd (perhaps we needed some safeguards and clearer errors in segment.py), but probably comes from overrunning the file if it was in fact an integer and you modified it to expect a float.  I think there are a few actions we need out of this:

1) extractions.py should be able to handle integers
2) check about another safeguard in segment.py, that ensures lines fall within the img_mm.shape[0] range.

However, I highly suggest that for your example you investigate the radiance that you have.  It sounds like you're using the AVIRIS-CL radiance that has not been converted to the appropriate units (which is where the int -> float change also happens in the AVIRIS workflow).  This isn't really an isofit issue, so for more here I suggest you contact the AVIRIS team for more details (and/or me, but let's do it on a different thread!).

evan-greenbrg, 2026-09-08:
Closing for limited traffic. Please reopen if any further issues come up.

closed 2026-09-08 by evan-greenbrg