isofit/isofit #312: Segmentation image not found

adamchlus, 2022-11-15:
When running `apply_oe.py` with the analytical line method and a non-standard sensor as input the script crashes because the segmentation file cannot be found:

`spectral.io.spyfile.FileNotFoundError: Unable to locate file "../output/PRS_20210317012458_20210317012502_0001_rdn_lbl.hdr". If the file exists, use its full path or place its directory in the SPECTRAL_DATA environment variable.`

However the segmentation image and header exist but are named differently:
```
../output/PRS_20210317012458_20210317012502_0001_lbl_lbl
../output/PRS_20210317012458_20210317012502_0001_lbl_lbl.hdr

pgbrodrick, 2022-11-15:
Hi Adam,

I think I see the inconsistency here, it's because I just do a substitution inside of the analytical_line, rather then passing in the label file.

Can you provide the full apply_oe.py call you use so I can confirm?

evan-greenbrg, 2026-09-08:
We do allow a segmentation file to be directly specified in the analytical line utility. However, apply OE will still expect a _lbl file with a specific naming convention to be present. The _lbl file should exist because we are always running the segmentation In apply OE.

```python
    lbl_file = (
        segmentation_file
        if segmentation_file
        else (subs_state_file.replace("_subs_state", "_lbl"))
    )
```

Tagging this as I/O related because we should potentially be able to support _lbl files (with different names) at the ApplyOE level.

closed 2026-09-08 by evan-greenbrg