isofit/isofit #314: Official Docker Image

jammont, 2022-11-28:
ISOFIT should have an official Docker image maintained by ISOFIT itself. This would come with all of the common dependencies of ISOFIT such as 6S and sRTMnet already installed and configured so that a user could just call the container and everything works. Additionally, this would streamline much of our testing environments such as CI/CD as to not need to reinstall ISOFIT for every PR/push.

Along with this, with proper documentation we can make it configurable for external emulators or other use cases. 

Altogether this should help streamline the installation process for common use cases.

pgbrodrick, 2023-01-24:
@serbinsh - would you mind taking a look at #321 and see how close this gets you (it's a general advance, but doesn't take into account anything specific for the apple silicon).  We're open to any input!

We will get the docker image hosted somewhere and referenced and add said image to the documentation before this issue is fully closed out.

serbinsh, 2023-01-24:
@pgbrodrick I reviewed and agree its an advance! I need to try running it as a Docker image and let you know what I learn but it does seem like a great step forward! Let me know where you end up hosting it for download OR I could just do a local build

That said I may also experiment with trying to make one that is dual AMD/ARM - I think it will just require finding a base image that has dual support.  I am assuming "mambaorg/micromamba:latest" is based on a debian or similar OS and may come prepackaged as AMD, ARM or both?  I would need to investigate.

Otherwise we would have to start from a dual OS then build on the mamba and python support before loading the other packages. 

I may also experiment with being able to also include libradtran as I have before but that may require more libs and may not be something relevant for most.  Also happy to just discuss more with you when convienient

pgbrodrick, 2023-03-14:
Hey all - I think there are some additional changes here thanks to @jammont, in particular with #313 updates to use the docker image and some additional changes to what's been uploaded.  We can continue to improve (and the docker image should be moved to the isofit org as soon as we can), but the base functionality is here.

@serbinsh / others, please feel free to open a new issue if we need additional functionality.