isofit/isofit #324: Wavelength consistency checking for inputs

davidraythompson, 2023-01-16:
I accidentally passed a "reference_reflectance_file" in the input block which had not been resampled to instrument wavelengths.  This caused a division operation with vectors of mismatched dimension.  The resulting exception was never printed to screen, producing a silent crash.  This might be another thing to watch for during user input checking.

unbohn, 2023-01-23:
Thanks for bringing this up, David! The new draft PR #322 adds a multitude of updates to the multisurface/macro-config approach also includes a check for wavelength consistency in the uses-provided input (see [here](https://github.com/unbohn/isofit_build_workflow/blob/cedff5469dae50abe081febcdb2916cc6d365593/isofit/utils/multisurface_oe.py#L140)). Right now, the code simply grabs the center wavelength from the radiance cube if either no wavelength file is provided or numbers of channels are not aligned. Additionally, we could of course throw a warning here. This absolutely might be something we want to include in `apply_oe.py` as well, so I could initiate this if desired.

pgbrodrick, 2024-09-12:
May be resolved?  @jammont, can you take a look?

unbohn, 2024-09-12:
@pgbrodrick I think this should be resolved. However, a second look by @jammont wouldn't hurt.

jammont, 2024-09-13:
Looks like this is still possible:
```python
# fileio.py L701
            if "radiometry_correction_file" in self.output_datasets:
                factors = np.ones(len(self.meas_wl))
                if "reference_reflectance_file" in self.input_datasets:
                    meas_est = fm.calc_meas(state_est, geom, rfl=reference_reflectance)

# forward.py L235
    def calc_meas(self, x, geom, rfl=None, Ls=None):
        x_surface, x_RT, x_instrument = self.unpack(x)
        rdn_hi = self.calc_rdn(x, geom, rfl, Ls)

# forward.py L219
    def calc_rdn(self, x, geom, rfl=None, Ls=None):
        x_surface, x_RT, x_instrument = self.unpack(x)
        if rfl is None:
            rfl = self.surface.calc_rfl(x_surface, geom)
        if Ls is None:
            Ls = self.surface.calc_Ls(x_surface, geom)

        rfl_hi = self.upsample(self.surface.wl, rfl)

#forward.py L363
    def upsample(self, wl, q):
        # In some cases, these differ only by a tiny amount,
        # so no need to waste time interpolating
        if (len(wl) == len(self.RT.wl)) and np.allclose(wl, self.RT.wl):
            return q

        if q.ndim > 1:
            q2 = []
            for qi in q:
                p = interp1d(wl, qi, fill_value="extrapolate")
                q2.append(p(self.RT.wl))
            return np.array(q2)
        else:
            p = interp1d(wl, q, fill_value="extrapolate")
            return p(self.RT.wl)
```
`fileio` will read in the reference reflectance and pass it to `forward.py` directly. Forward will attempt to interpolate it via the upsample function if the `self.surface.wl` and `fm.wl` don't match, but that doesn't consider the reference's shape. 

Should the upsample function attempt to correct any input that doesn't match by modifying to something like
```python
    def upsample(self, wl, q):
        """Linear interpolation to RT wavelengths."""
        # Only interpolate if these aren't close
        close = len(wl) == len(self.RT.wl) and np.allclose(wl, self.RT.wl)

        # or if any dimension is the wrong size
        interp = (np.array(q.shape) == len(self.RT.wl)).all()

        if not close or interp:
            if q.ndim > 1:
                for qi in q:
                    p = interp1d(wl, qi, fill_value="extrapolate")
                    q2.append(p(self.RT.wl))
                return np.array(q2)
            else:
                p = interp1d(wl, q, fill_value="extrapolate")
                return p(self.RT.wl)
        return q
```
or should it raise an exception instead, forcing the user to ensure the input reference matches? If an exception, then I propose enforcing that within `Forward.calc_meas`:
```python
    def calc_meas(self, x, geom, rfl=None, Ls=None):
        if not (np.array(rfl.shape) == len(self.RT.wl)).all():
            raise AttributeError(f'Reflectance does not match shape: expected {len(self.RT.wl)} got {q.shape}')
```

unbohn, 2024-09-14:
Thanks for looking into this, @jammont!  It looks indeed like that the current implementation of the upsample function doesn't check for consistency between the reference reflectance and both the surface and RT wavelengths. My suggestion would be to do the interpolation when the shapes of `q` and `self.RT.wl` are the same, but the values differ, and raise an exception if the shapes don't match. However, @pgbrodrick and @davidraythompson might have more thoughts on this.

evan-greenbrg, 2026-09-08:
Closed with #322  and #538

closed 2026-09-08 by evan-greenbrg