isofit/isofit #329: grpc thread pool hanging on fork

unbohn, 2023-02-06:
gRPC Python released version `1.52.0` of the `grpcio` package this morning, which seems to be incompatible with `ray = 2.2.0` (see [this grpc issue](https://github.com/grpc/grpc/issues/31885)). As a consequence, all our workflows that are using ray get stuck with the following error
`E0206 22:14:24.425686171 604 thread_pool.cc:254] Waiting for thread pool to idle before forking`. For now, the most promising solution would be to fix `grpcio` to `<1.52.0` in our environment file. However, further some testing is needed to evaluate how stable we are with this fix.

gnossen, 2023-02-06:
Please note that `grpcio==1.52.0` has been yanked.

unbohn, 2023-02-06:
Thanks, @gnossen! Looking forward to testing the upcoming `1.52.1` release soon.

pgbrodrick, 2023-07-24:
Seems resolved now, closing!