isofit/isofit #360: Import of ray in __init__ prevents automatic conda-forge upload.

unbohn, 2023-04-18:
With the implementation of the new ray wrapper, `ray` itself is now imported in the ISOFIT `__init__` module. However, this leads to an issue with the conda-forge package upload. The feedstock creates the upload distribution based on `setup.py`, which imports `__version__` from `__init__`. As `__init__` is called, it also tries to import `ray`, which is not available on conda and therefore, not installable via the conda-forge recipe. The recipe is only allowed to include packages, which are available through conda. As a consequence, the virtual environment on Azure doesn't provide the `ray` package, which leads to a pipeline failure, and prevents the automatic upload.

For now, I merged the upload PR manually into the feedstock, but it would be nice to retain the auto upload functionality. Two potential solutions come to my mind:

1. Creating a separate version file outside of `__init__`, so that the code is not attempting to import `ray` while executing `setup.py.`
2. Removing the import of `ray` from `__init__`. This is much less desirable since I don't have an alternative solution at hand right now.

I'm completely open to any thoughts here!

unbohn, 2023-04-19:
Ok, merging manually didn't lead to an upload of the new version to the conda-forge channel. The upload depends on a successful build, which is in turn depending on a succeeding Azure pipeline. So for now, we don't have version `2.9.9.2` on conda-forge.

jammont, 2023-05-24:
This issue was resolved by moving the version to `setup.cfg` in PR #342.

closed 2023-05-24 by jammont