isofit/isofit #365: install tensorflow version failure

pgbrodrick, 2023-06-05:
I'm getting a failure from a fresh mamba-based install:

Pip subprocess error:
ERROR: Could not find a version that satisfies the requirement tensorflow>2.0.1 (from versions: 2.13.0rc0, 2.13.0rc1)
ERROR: No matching distribution found for tensorflow>2.0.1

(from python 3.10.11).  We may need to update the environment_isofit_basic.yml file.

pgbrodrick, 2023-06-05:
Looking a little deeper, and this seems like this is due to ARM architecture on a mac.  Is there a reason we're using a pip instead of a mamba install of tensorflow?

jammont, 2023-06-05:
You are correct this is related to the [Mac M1s](https://stackoverflow.com/questions/72964800/what-is-the-proper-way-to-install-tensorflow-on-apple-m1-in-2022). The reason we use pip for Tensorflow is because it is recommended themselves:
> Do not install TensorFlow with conda. It may not have the latest stable version. pip is recommended since TensorFlow is only officially released to PyPI.

[Source](https://www.tensorflow.org/install). However, they also include this note for the Mac M1s:

> Note: For users of Apple M1 computers, to get native performance, you'll want to follow the instructions found [here](https://developer.apple.com/metal/tensorflow-plugin/). Conda has shown to have the smoothest install. Packages that include custom C++ extensions for TensorFlow also need to be compiled for Apple M1. Some packages, like [tensorflow_decision_forests](https://www.tensorflow.org/decision_forests) publish M1-compatible versions, but many packages don't. To use those libraries, you will have to use TensorFlow with x86 emulation and Rosetta.

This leads to instructions to install `tensorflow-macos` and `tensorflow-metal`. The StackOverflow link from above goes over this and some other information, so I'll look into this a bit more and test a solution soon.

PR #366 (2023-06-26): ARM64 Installation
# Description

This PR quickly resolves #365. 

# Solution

`tensorflow` on Mac M1s (ARM64 architecture) requires alternative tensorflow packages:
```yaml
conda channel: apple
  tensorflow-deps==2.9.0
pip:
  tensorflow-macos==2.9.2
  tensorflow-metal==0.5.1
```
`-macos` includes the `tensorflow` package and as such should not `pip install tensorflow`. Therefore, the `recipe/environment_isofit_basic.yml` has been duplicated for an M1 version which replaces the `tensorflow` package with the above. Most users will continue to use the basic.yml to install ISOFIT, while M1 users  [... cut]
closed 2023-07-04 by pgbrodrick