isofit/isofit #393: ISOFIT installation to include extra data files

jammont, 2023-10-02:
We rely on a number of in-house data files to be available when executing the core of ISOFIT (such as the earth-sun distance file). We should double check to ensure these files are being included in all available distributions of ISOFIT. Presently, we know these will work if the repo is pulled, but it needs to be confirmed for the following as well:

- [x] github
- [ ] pypi
- [ ] anaconda

danschef, 2023-10-18:
During some tests together with @unbohn at the GFZ, it turned out that at least the `data` and `examples` folders are missing. We only tested the latest conda-forge build but I guess it is the same issue for PyPI. Might be something in the [MANIFEST.in](https://github.com/isofit/isofit/blob/main/MANIFEST.in).

unbohn, 2023-10-18:
Thanks, @jammont and @danschef! I'm aware of this issue and will work and on a solution.

unbohn, 2023-12-10:
@pgbrodrick @jammont: I made some comprehensive tests and can confirm that both `data` and `examples` directories are  currently missing in both conda and pip installations of ISOFIT. The problem is that those directories are sitting one level up from the actual package directory in the repository. 

According to the [setuptools userguide](https://setuptools.pypa.io/en/latest/userguide/miscellaneous.html), only files inside the package directory are included in the final wheel. When included through the `MANIFEST.in`, directories outside of the package folder will be present in the sdist but not in the wheel. The Python Packaging Authority (PPA) actually recommends that any data files we wish to be accessible at run time be included inside the package (see the [data files support by setuptools](https://setuptools.pypa.io/en/latest/userguide/datafiles.html)).

Question: Do we want to move the `data` and `examples` directories into the package directory? We'd need to slightly modify the `MANIFEST.in` and would then have both directories available at the `site-packages` directory when running `conda install isofit` or `pip install isofit`. Of course, we'd need to modify some path inquiries in the code, in particular for accessing files in the `data` directory, but that should be easy to implement.

pl-kevinwurster, 2023-12-11:
Tracking this ticket so that I can figure out how to test in https://github.com/isofit/isofit/pull/421.

I think a bit of work will also be required in order to read data files using canonical tooling. This section in the `setuptools` docs describes how to do so:

https://setuptools.pypa.io/en/latest/userguide/datafiles.html#accessing-data-files-at-runtime

Using module `__file__` attributes to find neighboring files is acceptable, but will not work for Egg or zipfile installs.

----

I would like to push back on "just" including the `data/` and `examples/` directories though. They are quite large – especially `examples/`:

```console
$ du -sh data examples
 28M    data
101M    examples
```

So, some questions:

1. Are the entire contents of `data/` required for `isofit`?
2. Are the contents of `data/` required to make `isofit` work at all, or only specific features?
3. Why is sRTM Net not included in `data`?
4. What is the purpose of including `examples/`? What problem does this solve?

unbohn, 2023-12-11:
@pl-kevinwurster Thanks for pointing to the `setuptools` docs! We'll work on replacing module `__file__` attributes by `importlib.resources` to access files needed at runtime.

We don't plan to include the entire `examples/` directory, but just a selection of files from the `20171108_Pasadena` one, as you can see in the `MANIFEST.in`. They should be quite smaller than 101M.

Now some answers to your questions:

1./2. The contents of `data/` are not required to make `isofit` work. However, they include instrument specific noise files, aerosol templates, earth-sun distances, and solar irradiance models. Using those improves the accuracy of `isofit` retrievals, so would be nice to have for users. Other features such as the EWT retrieval (which is decoupled from `isofit` though) depend on files in the `data/` directory. Finally, it also includes surface reflectance spectral libraries that can be used to build the surface model in cases where users don't provide their own model. If we don't include the `data/` directory, we'd need to modify `apply_oe` a little bit since in currently tries to obtain a few files without having a backup, such as an `if...else` or `try:...except` clause.

3. I assume that sRTMnet is simply too large to be included in the `data/` directory, but I'm not fully sure.

4. Including files from the `examples/` directory provides some basic info on how to run `isofit`.

In any case, completely open to thoughts here!

unbohn, 2023-12-18:
@pl-kevinwurster Thanks for this comprehensive feedback! I try to respond to your points below:

- I think the driving thought for including the `examples` directory in the package was that `apply_oe.py` pulls a data file from that directory [here](https://github.com/isofit/isofit/blob/1818ef6fbdcc88c80e72f3533602410c86ecfd8b/isofit/utils/apply_oe.py#L760C5-L760C5). However, I fully agree that this is a suboptimal solution and will work on setting up an alternative approach of providing files needed at runtime.
- I finally agree that we shouldn't store any contents of the `examples` directory in the package upload. Using an extended documentation and/or a separate repository sounds great to me.
- The same applies to the `data` directory. I'll work on getting all files needed at runtime to be stored in a more clever way, without the need to add an external directory to the package.
- It certainly makes sense to think through versioning of data files, different functionalities of `isofit`, and sensor-specific files.
- I think the users shouldn't have to deal with also configuring data files as you're right that `isofit` requires a relatively comprehensive configuration setting.
- The inclusion of the `data` directory is not driven by the interest of installing `isofit` via `PYTHONPATH`.
- I like your ideas of both adding data packages and moving data dependencies to a separate GitHub repository. Let's discuss this in a larger group.

PR #448 (2024-02-27): CLI for downloading extras
# Description
This PR implements the capability to download extra files that are useful for ISOFIT. This functionality will re-enable support for testing our examples in the new workflows (#421) as well as resolve #393.

# Changes
- Adds the subcommand `isofit download` which houses other subcommands:
  - `data` to download data files from https://github.com/isofit/isofit-data
  - `examples` to download from https://github.com/isofit/isofit-tutorials

# todo
- Expand commands to include installation of:
  - [ ] sRTMnet
  - [ ] 6S
closed 2024-02-27 by unbohn