isofit/isofit #411: Spring cleaning of package dependencies

davidraythompson, 2023-11-20:
There are many dependencies (gdal, matplotlib, pandas, etc.) that are not really needed.  We should simplify the code.

pl-kevinwurster, 2023-11-20:
Some recommendations on installation dependencies based on my own observations, and use. Loosely ranked from "yes easy", to "hard", and "I'm not sure if this is the right choice for the project".

| Package | Suggested Change | Notes |
|:-------:|:----------------:|:------|
| `matplotlib` | Remove | Imported in `isofit/utils/empirical_line.py`, but is only used in a single "private" `_plot_example()` debug function. |
| `pep8` | Remove | Appropriate for development, but not a runtime requirement. |
| `pytest` | Remove | Appropriate for development, but not a runtime requirement. |
| `click` | `click>=4.0` | Based on my own experience using click in a more modern dependency tree. |
| `protobuf` | Remove | Handled by https://github.com/isofit/isofit/pull/398. |
| `gdal` | Remove | Mostly, `isofit` uses `spectral` for reading ENVI files, except when it doesn't. I do see some instances of using GDAL to read metadata, but I think these are ENVI files, so `spectral` could still be used? |
| `pandas` | Remove | Used in `.imgspec` and `isofit/inversion/inverse_simple.py` to read CSVs and an Excel file. Use seems to be limited to reading data into 2D arrays, so invocations could likely be replaced with a bit of code, and the `csv` library? Pandas doesn't (currently) have any difficult dependencies, but it does increase the dependency surface area to support. |
| `Numpy` | `numpy>=1.17` | [Python 3.8 is officially supported by the Python Software Foundation until the end of 2024.](https://devguide.python.org/versions/) Numpy 1.17 is the oldest version compatible with Pytho 3.8. IMO both of these are reasonable to support, but does constrain `isofit`'s Numpy use to a subset of the API. The [`oldest-supported-numpy`](https://pypi.org/project/oldest-supported-numpy/) package may be useful here. |

pgbrodrick, 2024-05-13:
Marking as resolved - pandas dependency still holds.

@jammont , can you comment on the pre-commit dependencies?

jammont, 2024-05-13:
Pep8 testing is disabled at this time.

For `isort` and `black`, the `.pre-commit-config.yaml` file simply installs these in an environment by running `pre-commit install`. These tools are executed when a user creates a commit. This has been working fine, in my experience. Can it be further commented that these are not working?

If installed correctly, you should see:
```bash
$ git commit -m "..."         
isort....................................................................Passed
black....................................................................Passed
```
```bash
$ git commit -m "..."     
isort....................................................................Passed
black....................................................................Failed
- hook id: black
- files were modified by this hook

reformatted isofit/radiative_transfer/engines/templates/full.py
```

brentwilder, 2026-09-08:
As of 8 September, 2026, the following dependencies have been removed: `gdal`, `matplotlib`, `pandas`. The `pandas` dependency was removed more recently in #759 .

dependencies = [
  "click",
  "dask",
  "h5py <= 3.14.0",
  "netCDF4 < 1.7.4",
  "numba",
  "numpy >= 1.20",
  "torch",
  "pyyaml >= 5.3.2",
  "ray >= 1.2.0",
  "scikit-image >= 0.17.0",
  "scikit-learn >= 0.19.1",
  "scipy >= 1.3.0",
  "spectral >= 0.19",
  "xarray",
  "xxhash >= 1.2.0",
  "zarr >= 3",
]

We currently use all packages above in some fashion throughout the codebase. We should continue to revisit this periodically to ensure we are keeping the isofit core dependencies lightweight.

closed 2026-09-08 by brentwilder