isofit/isofit #412: Incomplete LUT grid in apply_oe when all z values in loc file are < 0 (when using emulator)

NickVaughn, 2023-11-28:
I have a couple flight lines where z values are all slightly below 0.  These all get converted to 0 for 6S, but this results in no range in elevation to build out the LUT grid for GNDALT and it only has one entry.  Maybe this could be fixed with a quick check could be implemented to check for this condition, and add a second entry with a little bit of elevation added? 

```
INFO:2023-11-27,17:52:20 ||| Scene contains target lut grid elements < 0 km, and uses 6s (via sRTMnet).  6s does not support targets below sea level in km units.  Setting grid points [-3.8420957e+03 -3.8418457e+03 -3.8415957e+03 ... -5.0370000e-01
 -2.5370000e-01 -3.7000000e-03] to 0.
INFO:2023-11-27,17:52:20 ||| Scene contains a mean target elevation < 0.  6s does not support targets below sea level in km units.  Setting mean elevation to 0.
...
INFO:2023-11-27,17:52:20 ||| Running ISOFIT with full LUT
INFO:2023-11-27,17:52:20 ||| Checking config sections for configuration issues
ERROR:2023-11-27,17:52:20 ||| lut_grid item GNDALT has less than the required 2 elements
Traceback (most recent call last):
  File "/home/nvaughn4/repos/isofit/isofit/utils/apply_oe.py", line 1978, in <module>
    _cli()
  File "/home/nvaughn4/bin/miniconda3/envs/isofit_env/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/nvaughn4/bin/miniconda3/envs/isofit_env/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/nvaughn4/bin/miniconda3/envs/isofit_env/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/nvaughn4/bin/miniconda3/envs/isofit_env/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/nvaughn4/repos/isofit/isofit/utils/apply_oe.py", line 1972, in _cli
    apply_oe(SimpleNamespace(**kwargs))
  File "/home/nvaughn4/repos/isofit/isofit/utils/apply_oe.py", line 495, in apply_oe
    retrieval_full = isofit.Isofit(
  File "/home/nvaughn4/repos/isofit/isofit/core/isofit.py", line 68, in __init__
    self.config.get_config_errors()
  File "/home/nvaughn4/repos/isofit/isofit/configs/configs.py", line 115, in get_config_errors
    raise AttributeError("Configuration error(s) found.  See log for details.")
AttributeError: Configuration error(s) found.  See log for details.
isofit returned non-zero code (1) - aborting
srun: error: cg17-15: task 0: Exited with exit code 1
```

Config has this for lut_grid:

```
...
        "radiative_transfer": {
            "lut_grid": {
                "AOT550": [
                    0.001,
                    0.1009,
                    0.2008,
                    0.3007,
                    0.4006,
                    0.5005,
                    0.6004,
                    0.7003,
                    0.8002,
                    0.9001,
                    1.0
                ],
                "GNDALT": [
                    0.0
                ],
                "H2OSTR": [
                    0.4332,
                    0.6653,
                    0.8975,
                    1.1296,
                    1.3618,
                    1.5939,
                    1.8261,
                    2.0582,
                    2.2904,
                    2.5225,
                    2.7547,
                    2.9868,
                    3.219,
                    3.4511
                ],
                "OBSZEN": [
                    159.0915,
                    168.1737,
                    177.256
                ],
...
```

pgbrodrick, 2023-11-28:
Can you say a bit more about the style of use here?

The current assumed behavior is that LUT grids have at least two dimensions (otherwise it's not really a grid...).  In the case of the above, the simple solution, which I believe is followed by our utils scripts, is to just not include the GNDALT dimension in the config.  In general we prefer this for the LUT to keep the dimensionality low.

We could implement a simple check for this condition in the config that pops out 1-D LUT elements, but that would risk a deviation between the LUT grid and the RTE template file.  We could add additional checks to force these to be the same, but that starts to be a decent amount of checking for a case that's already covered.

If the above solution of just trimming out GNDALT from the config doesn't suffice, maybe you could say a bit more about the desired use case here to help guide ideas?

pgbrodrick, 2023-11-30:
Ahh, okay, I had misunderstood.

Looks like there's a super simple fix here.  We just need to add a quick check on the elevation grid being length 0, and if it is re-assign for both the template and set the grid to null.  Standby for PR...

PR #415 (2023-12-04): Length 1 elevation LUT protection
Small patch to resolve #412.
closed 2023-12-04 by unbohn