isofit/isofit #417: IndexError during LUT interpolation

davidraythompson, 2023-12-01:
Running with a small (2-dimensional lookup table) I get the following error: 

Traceback (most recent call last):
  File "/tmp/miniconda/envs/isofit_env/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/tmp/miniconda/envs/isofit_env/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/drt/src/isofit/isofit/__main__.py", line 3, in <module>
    cli()
  File "/tmp/miniconda/envs/isofit_env/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/tmp/miniconda/envs/isofit_env/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/tmp/miniconda/envs/isofit_env/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/tmp/miniconda/envs/isofit_env/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/tmp/miniconda/envs/isofit_env/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/drt/src/isofit/isofit/core/isofit.py", line 314, in _cli
    Isofit(config_file=config_file, level=level).run()
  File "/home/drt/src/isofit/isofit/core/isofit.py", line 180, in run
    self.workers.map_unordered(
  File "/home/drt/src/isofit/isofit/wrappers/ray.py", line 91, in map_unordered
    return [func(*pair).get() for pair in zip(self.actors, iterable)]
  File "/home/drt/src/isofit/isofit/wrappers/ray.py", line 91, in <listcomp>
    return [func(*pair).get() for pair in zip(self.actors, iterable)]
  File "/home/drt/src/isofit/isofit/wrappers/ray.py", line 39, in get
    return self.obj(*self.args, **self.kwargs)
  File "/home/drt/src/isofit/isofit/core/isofit.py", line 256, in run_set_of_spectra
    states = self.iv.invert(input_data.meas, input_data.geom)
  File "/home/drt/src/isofit/isofit/inversion/inverse.py", line 340, in invert
    x0 = invert_simple(self.fm, meas, geom)
  File "/home/drt/src/isofit/isofit/inversion/inverse_simple.py", line 346, in invert_simple
    x[forward.idx_RT] = heuristic_atmosphere(
  File "/home/drt/src/isofit/isofit/inversion/inverse_simple.py", line 108, in heuristic_atmosphere
    rhi = RT.get_shared_rtm_quantities(x_RT_2, geom)
  File "/home/drt/src/isofit/isofit/radiative_transfer/radiative_transfer.py", line 131, in get_shared_rtm_quantities
    ret.append(RT.get(x_RT, geom))
  File "/home/drt/src/isofit/isofit/radiative_transfer/sRTMnet.py", line 338, in get
    return self._lookup_lut(point)
  File "/home/drt/src/isofit/isofit/radiative_transfer/sRTMnet.py", line 295, in _lookup_lut
    ret = {key: lut(point) for key, lut in self.luts.items()}
  File "/home/drt/src/isofit/isofit/radiative_transfer/sRTMnet.py", line 295, in <dictcomp>
    ret = {key: lut(point) for key, lut in self.luts.items()}
  File "/home/drt/src/isofit/isofit/core/common.py", line 260, in __call__
    return self._multilinear_grid(*args, **kwargs)
  File "/home/drt/src/isofit/isofit/core/common.py", line 213, in _multilinear_grid
    inds = [
  File "/home/drt/src/isofit/isofit/core/common.py", line 214, in <listcomp>
    np.searchsorted(t[:-1], points[i]) - 1
IndexError: index 2 is out of bounds for axis 0 with size 2

For what it's worth, here is my config file:
{
    "ISOFIT_base": "/home/drt/src/isofit",
    "forward_model": {
        "instrument": {
            "SNR": 500,
            "integrations": 100,
            "unknowns": {
                "uncorrelated_radiometric_uncertainty": 0.01
            },
            "wavelength_file": "/beegfs/scratch/drt/20231126_ANG_BioSCapeVicariousCal/data/wavelengths.txt"
        },
        "radiative_transfer": {
            "lut_grid": {
                "AOT550": [
                    0.001,
                    0.1,
                    0.2
                ],
                "H2OSTR": [
                    1.1995,
                    1.448,
                    1.6966,
                    1.9452,
                    2.1937,
                    2.4423,
                    2.6908,
                    2.9394
                ]
            },
            "radiative_transfer_engines": {
                "vswir": {
                    "aerosol_model_file": "/home/drt/src/isofit/data/aerosol_model.txt",
                    "aerosol_template_file": "/home/drt/src/isofit/data/aerosol_template.json",
                    "earth_sun_distance_file": "/home/drt/src/isofit/data/earth_sun_distance.txt",
                    "emulator_aux_file": "/beegfs/scratch/brodrick/emulator/sixs-modtran-emulator/hialt_reruns/tenth_mse_dense/emulator_random_aux.npz",
                    "emulator_file": "/beegfs/scratch/brodrick/emulator/sixs-modtran-emulator/hialt_reruns/tenth_mse_dense/emulator_random.h5",
                    "engine_base_dir": "/home/drt/src/6sv/",
                    "engine_name": "sRTMnet",
                    "interpolator_base_path": "/beegfs/scratch/drt/20230216_SMODE_InsituCal/rfl/prm20221024t195127/lut_full/emulator_random_vi",
                    "irradiance_file": "/home/drt/src/isofit/examples/20151026_SantaMonica/data/prism_optimized_irr.dat",
                    "lut_names": [
                        "H2OSTR",
                        "AOT550"
                    ],
                    "lut_path": "/beegfs/scratch/drt/20231126_ANG_BioSCapeVicariousCal/lut_full",
                    "statevector_names": [
                        "H2OSTR",
                        "AOT550"
                    ],
                    "template_file": "/beegfs/scratch/drt/20231126_ANG_BioSCapeVicariousCal/config/ang20231113t134459_modtran.json"
                }
            },
            "statevector": {
                "AOT550": {
                    "bounds": [
                        0.001,
                        0.2
                    ],
                    "init": 0.1009,
                    "prior_mean": 0.1009,
                    "prior_sigma": 10.0,
                    "scale": 1
                },
                "H2OSTR": {
                    "bounds": [
                        1.1995,
                        2.9394
                    ],
                    "init": 2.1936999999999998,
                    "prior_mean": 2.1936999999999998,
                    "prior_sigma": 100.0,
                    "scale": 1
                }
            },
            "unknowns": {
                "H2O_ABSCO": 0.0
            }
        },
        "surface": {
            "select_on_init": true,
            "surface_category": "multicomponent_surface",
            "surface_file": "/beegfs/scratch/drt/20231126_ANG_BioSCapeVicariousCal/data/surface.mat"
        }
    },
    "implementation": {
        "debug_mode": false,
        "inversion": {
            "windows": [
                [
                    350.0,
                    1360.0
                ],
                [
                    1410,
                    1800.0
                ],
                [
                    1970.0,
                    2500.0
                ]
            ]
        },
        "n_cores": 40,
        "ray_temp_dir": "/tmp/ray"
    },
    "input": {
        "measured_radiance_file": "/beegfs/scratch/drt/20231126_ANG_BioSCapeVicariousCal/data/ang20231113t134459_site0_original_rdn.txt"
    },
    "output": {
        "estimated_reflectance_file": "/beegfs/scratch/drt/20231126_ANG_BioSCapeVicariousCal/output/ang20231113t134459_site0_rfl.txt",
        "estimated_state_file": "/beegfs/scratch/drt/20231126_ANG_BioSCapeVicariousCal/output/ang20231113t134459_site0_state.txt"
    }
}

davidraythompson, 2023-12-06:
Debugging, it looks like some geometry parameters (OBSZEN and RELAZ from the ranges) were inserted automatically into the interpolator's gridtuples, causing an error.

pgbrodrick, 2024-05-13:
I expect this is resolved w/ 3.0 advances.  Marking as resolved - please re-open as needed.