isofit/isofit #418: CI and Developer Experience Improvements

pl-kevinwurster, 2023-12-05:
I am working on a set of CI and developer experience improvements for ISOFIT. The work is very tightly coupled, and rather than track each item across many tickets, I would like to track the work here.

At a high level the work focuses on two realms:

1. Provide developers with clear instructions for how to set up an environment, run tests, etc.
2. Directly test the `pip` package across many Python versions to provide a foundation for other packaging mechanisms, and confidence for users that the package can be installed 

The todo list:

- [ ] Clear instructions for setting up a developer environment. Ideally expressed as a shell script that is reasonably portable.
- [ ] A reasonable starting point for users to understand how to `$ pip install isofit`. Ultimately this is extremely system and environment dependent, so there is no single right answer. At the very least I will provide some tips for getting the `gdal` Python package installed.
- [ ] Encapsulate sRTM net download/unpack in a shell script.
- [ ] Encapsulate 6S download/build in a shell script.
- [ ] Provide a mechanism for running tests with sRTM net and 6S environment variables set.
- [ ] ISOFIT requires `pygrib`, but does not list it as a project dependency.
- [ ] GitHub Action for testing the `pip` package on all supported versions of Python.
- [ ] GitHub Action for ensuring installing via `PYTHONPATH` works correctly (related to https://github.com/isofit/isofit/issues/413).
- [ ] Fix or remove `pep8` checks. Currently implemented as a test in the test suite, but is not actually doing anything.

pgbrodrick, 2024-05-13:
#452 is now resolved, making the lower level to-do complete.  The high level pieces still exist.  To make more tractable, I suggest breaking out into individual issues.

pl-kevinwurster, 2024-05-14:
@pgbrodrick Closing this ticket. My initial pain points have been addressed.

closed 2024-05-14 by pl-kevinwurster