isofit/isofit #444: Do not set environment variables

pl-kevinwurster, 2024-02-23:
I have found 3 instances of ISOFIT settings environment variables inside the Python library. This is generally not acceptable because users employing ISOFIT combined with other tooling will see behavior changes caused by ISOFIT setting these variables. For example, just importing `isofit.core.luts` will disable HDF5 locking for *ANY* `h5py` code operating in the same scope. In code terms, the `h5py.File()` call will have locking disabled:

```python
import isofit.core.luts
import h5py

# Not actually interacting with 'isofit', but locking is disabled for the line below.
with h5py.File('data.h5', 'r') as f:
    ...
```

* [`HDF5_USE_FILE_LOCKING`](https://github.com/isofit/isofit/blob/5cf282b7a8f58c9b89e8d580ec0db54e9ae9ddc6/isofit/radiative_transfer/luts.py#L13)
* [`MKL_NUM_THREADS`](https://github.com/isofit/isofit/blob/5cf282b7a8f58c9b89e8d580ec0db54e9ae9ddc6/isofit/core/isofit.py#L51)
* [`OMP_NUM_THREADS`](https://github.com/isofit/isofit/blob/5cf282b7a8f58c9b89e8d580ec0db54e9ae9ddc6/isofit/core/isofit.py#L52C21-L52C36)

pgbrodrick, 2024-02-24:
Thanks for opening this.

@jammont, I believe we can use the locking parameter instead of the environment variable for HDF5_USE_FILE_LOCKING (see https://docs.h5py.org/en/stable/high/file.html#reference).

MKL and OMP num threads are trickier because of the number of modules that may using threading under the hood.  Here are some viable options:
- Purge entirely and require users to set these parameters - but _most_ users will want these set and would have to thoroughly read documentation to know they need to be set
- Migrate the MKL and OMP environment changes to apply_oe.  The idea being anyone using apply_oe will want this, and it is less likely - thought not impossible - for apply_oe to be called in the same scope as other code.  However, this wouldn't satisfy individual inversions....so for example, most tests would now need multiple environment calls to operate efficiently.
- Leave these two as-are and make their being set more explicit in documentation, understanding that these impact only performance.
- Comb through and attempt to set every packages thread limit on load.  From experience, this is hard to do.

Open to thoughts and suggestions!

pl-kevinwurster, 2024-02-26:
Ray has a mechanism for defining environment variables within the worker environment. I suspect that Ray is effectively doing something like:

jammont, 2024-03-21:
I believe we can remove `HDF5_USE_FILE_LOCKING`. This was implemented back when we were trying to stream multiple writers to the single netcdf in parallel which needed to disable the locking, but now that we lock the file anyways it is no longer relevant.

pgbrodrick, 2024-05-13:
- HDF5_USE_FILE_LOCKING is now removed.
- Suggestion is to check OMP / MKL environment values and raise a very strong warning.

PR #535 (2024-09-23): MKL/OMP env vars fixed
Fixed a longstanding bug where `MKL_NUM_THREADS` and `OMP_NUM_THREADS` was not being set properly by ISOFIT. These need to be set before any import/instantiation of numpy.

Additionally adds a new ISOFIT env var `ISOFIT_NO_SET_THREADS` to disable automatically setting these for advanced users.

Closes #444
closed 2024-09-23 by pgbrodrick