isofit/isofit #454: apply_oe non-run

pgbrodrick, 2024-02-29:
Currently, apply_oe.py (on dev) does not execute when called.  This must be a consequence of the recent click revision.  Noting here while we work on it.

pgbrodrick, 2024-02-29:
This is a consequence of removing the __main__ block.  Calls that previously looked like:

`python apply_oe.py .....`

should now look like:

`isofit applye_oe .....`

pl-kevinwurster, 2024-02-29:
@pgbrodrick Based on previous discussions, I'm guessing you are working with a version of ISOFIT installed via `$PYTHONPATH`. If that is the case, you probably want to use `$ python3 -m isofit apply_oe` rather than `$ isofit apply_oe`, as the `$ isofit` executable will not exist in this situation.

I included some details about this in the PR description for https://github.com/isofit/isofit/pull/422#issue-2029561159.

pgbrodrick, 2024-05-13:
This is resolved!

closed 2024-05-13 by pgbrodrick