isofit/isofit #466: More useful error messages if environment is not set up correctly or extra-files are missing

danschef, 2024-03-15:
isofit 3.0 (current dev branch)

There are a couple of not very user-friendly exceptions that are raised in case the environment is not correctly configured or the ISOFIT extra-files are missing. I think these cases could easily be catched beforehand and a useful error message could be raised pointing to the solution.

<details> 
  <summary>FileNotFoundError if `isofit download data` was not executed -> data folder missing</summary>
    
```python
test_isofit_enmap.py::Test_ISOFIT_EnMAP::test_run_isofit 

======================== 1 failed, 3 warnings in 30.74s ========================
2024-03-14 20:17:49,255	INFO worker.py:1724 -- Started a local Ray instance.
Stopping ray.
FAILED
tests/test_isofit_enmap.py:63 (Test_ISOFIT_EnMAP.test_run_isofit)
self = <tests.test_isofit_enmap.Test_ISOFIT_EnMAP testMethod=test_run_isofit>

    def test_run_isofit(self):
        # from geoarray import GeoArray
        # gA = GeoArray('/home/gfz-fe/scheffler/temp/EnPT/isofit_implementation/data_in/emp20220712t184754_rdn_sub.bil')
        # gA.show()
    
        from time import time
        t0 = time()
    
>       IsofitEnMAP()._apply_oe(
            # input_radiance='/home/gfz-fe/scheffler/temp/EnPT/isofit_implementation/data_in/emp20220712t184754_rdn_sub.bil',
            input_radiance='/home/gfz-fe/scheffler/temp/EnPT/isofit_implementation/data_in/ENMAP01-____L1X-DT000000XXXX_20220712T000000Z_00x_VXXXXXX_XXXXXXTXXXXXXZ.bil',
            input_loc='/home/gfz-fe/scheffler/temp/EnPT/isofit_implementation/data_in/emp20220712t184754_loc_sub.bil',
            input_obs='/home/gfz-fe/scheffler/temp/EnPT/isofit_implementation/data_in/emp20220712t184754_obs_sub.bil',
            working_directory='/home/gfz-fe/scheffler/temp/EnPT/isofit_implementation/data_out/',
            surface_path='/home/gfz-fe/scheffler/temp/EnPT/isofit_implementation/surface/surface_20221020_EnMAP.mat',
            wavelength_path='/home/gfz-fe/scheffler/temp/EnPT/isofit_implementation/sensor_new/enmap_wavelengths.txt',
            log_file='/home/gfz-fe/scheffler/temp/EnPT/isofit_implementation/data_out/isofit.log',
            presolve=True,
            emulator_base='/home/gfz-fe/scheffler/sRTMnet_v100/sRTMnet_v100',  # FIXME why not /home/gfz-fe/scheffler/sRTMnet_v100/
            n_cores=30
        )

test_isofit_enmap.py:72: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../enpt/processors/atmospheric_correction/_isofit_enmap.py:108: in _apply_oe
    apply_oe(SimpleNamespace(**params))
../../../../mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/isofit/utils/apply_oe.py:590: in apply_oe
    retrieval_full.run()
../../../../mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/isofit/core/isofit.py:182: in run
    res = list(
../../../../mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/ray/util/actor_pool.py:170: in get_generator
    yield self.get_next_unordered()
../../../../mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/ray/util/actor_pool.py:370: in get_next_unordered
    return ray.get(future)
../../../../mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/ray/_private/auto_init_hook.py:22: in auto_init_wrapper
    return fn(*args, **kwargs)
../../../../mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/ray/_private/client_mode_hook.py:103: in wrapper
    return func(*args, **kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

object_refs = [ObjectRef(8c4854248414f6339358be101ae77cb2af9663a30100000001000000)]

    @PublicAPI
    @client_mode_hook
    def get(
        object_refs: Union["ObjectRef[Any]", Sequence["ObjectRef[Any]"]],
        *,
        timeout: Optional[float] = None,
    ) -> Union[Any, List[Any]]:
        """Get a remote object or a list of remote objects from the object store.
    
        This method blocks until the object corresponding to the object ref is
        available in the local object store. If this object is not in the local
[... 222 lines cut ...]
  File "/home/gfz-fe/mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/isofit/radiative_transfer/radiative_transfer_engine.py", line 624, in streamSimulation
    data = reader(point)
           ^^^^^^^^^^^^^
  File "/home/gfz-fe/mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/isofit/radiative_transfer/modtran.py", line 270, in readSim
    solzen = self.load_tp6(f"{file}.tp6")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gfz-fe/mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/isofit/radiative_transfer/modtran.py", line 200, in load_tp6
    with open(file, "r") as tp6:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpfm0mwnjo/lut_full/AERFRAC_2-0.6004_H2OSTR-1.7825_surface_elevation_km-1.0528.tp6' at time: 1.71042e+09
:task_name:streamSimulation
:task_name:streamSimulation
[2024-03-14 13:50:38,885 E 1079104 1079104] core_worker.cc:1749: Pushed Error with JobID: 01000000 of type: task with message: ray::streamSimulation() (pid=1079104, ip=139.17.56.61)
  File "/home/gfz-fe/mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/isofit/radiative_transfer/radiative_transfer_engine.py", line 624, in streamSimulation
    data = reader(point)
           ^^^^^^^^^^^^^
  File "/home/gfz-fe/mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/isofit/radiative_transfer/modtran.py", line 270, in readSim
    solzen = self.load_tp6(f"{file}.tp6")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gfz-fe/mambaforge/envs/enpt_full_dev/lib/python3.11/site-packages/isofit/radiative_transfer/modtran.py", line 200, in load_tp6
    with open(file, "r") as tp6:
         ^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpfm0mwnjo/lut_full/AERFRAC_2-0.6004_H2OSTR-1.7825_surface_elevation_km-1.2602.tp6' at time: 1.71042e+09
```
</details>

danschef, 2024-03-18:
@unbohn

betsymfarris, 2024-04-11:
Files are missing for running examples via Quick Start instructions in v3.0.0., e.g:
cd examples/20171108_Pasadena
./run_examples_modtran.sh

PR #561 (2024-10-28): Helpful Validation
Attempting to provide more helpful debugging/error messages when ISOFIT is missing extra dependencies.

The `validate` functions have added two parameters: `debug` and `error`. These are callables for where to route messages to. By default they route to `print` which is used by the `isofit build` command as there's no logging. However, these change to `logging.debug` and `logging.error` when called from the new `path` function, which is used during an isofit pipeline execution (ie. `apply_oe` or `Isofit().run()`)

This ensures these messages get written to a log file if one  [... cut]
closed 2024-10-28 by pgbrodrick