isofit/isofit #510: Tutorial Questions

pgbrodrick, 2024-06-24:
Hey All,

What are the questions that we want to see folded into a tutorial series for ISOFIT?

barentsen, 2024-07-08:
Just to get started, below is an example list of bite-sized tutorials which could make ISOFIT more accessible to new users.

I came up with this list in 15 minutes, so don't take it too seriously. It has not been prioritized or well-thought through; I merely generated this list to kick-start brainstorming on this topic. I am aware that some of these topics are already covered in the existing documentation, but it seems fine for tutorials to overlap with more formal documentation resources. I am also aware that it would take a LOT of work to generate all these tutorials.

- Getting Started:
    - Quickstart: How to correct EMIT L1B data using default options?
    - Input Data: How to prepare input data for use in ISOFIT?
    - Output Data: How to analyze and interpret the results?
- Configuring ISOFIT: How to configure the forward model?
    - Config File: How is the configuration file structured?
    - Radiative Transfer Engines: How to pick and configure the RT engine?
    - Surface Priors: How to create a surface model?
    - Atmosphere Priors: How to configure the atmosphere parameters?
    - Aerosol Models: How to choose and configure aerosol options?
- Interpreting ISOFIT Output:
    - Visualizing Output: How to plot and inspect the reflectance values?
    - Uncertainty Analysis: How to inspect and interpret the uncertainties?
    - Parameter Sensitivity Analysis: How to understand the impact of different parameters?
- Common Problems:
    - NetCDF Support: How to prepare input data in NetCDF format? 
    - Optimizing Performance: How to speed up ISOFIT runs?
    - Cloud Masking: How to detect and handle clouds in the input data?
    - Data Validation: How to compare ISOFIT results against ground truth measurements?
- Advanced Use Cases:
    - Coastal and Aquatic Applications: How to use ISOFIT over water?
    - Glint Mode: How to use ISOFIT in the presence of specular reflections?
    - How to characterize wildfire smoke? (aka, How to reproduce Brodrick+ 2022?)
    - How to retrieve water content? (aka, How to reproduce Bohn+ papers?)
    - How to calibrate EMIT? (aka, How to reproduce Thompson+ 2024?)
    - Integrating ISOFIT with Python Scripts: How to use the Python API?
- New Missions:
    - How to run ISOFIT on data from new instruments?
    - Noise Models: How to create the instrument noise models?
    - Instrument Calibration: How to evaluate spectral and radiometric calibrations?
- Contributing to ISOFIT
    - Modifying Code: How to propose a new feature or bug fix?
    - Documentation: How to prepare a new tutorial?
    - Testing: How to run the examples and unit tests?

keely-roth, 2024-07-15:
Wow! now that's a list :) 
I think content-wise, I'd double tap these as essential, though I can see some ways to collate/collapse some of them together.

- Getting Started:
     - Quickstart: How to correct EMIT L1B data using default options?
     - Input Data: How to prepare input data for use in ISOFIT?
     - Output Data: How to analyze and interpret the results?
- Configuring ISOFIT: How to configure the forward model?
     - Config File: How is the configuration file structured?
     - Radiative Transfer Engines: How to pick and configure the RT engine?
     - Surface Priors: How to create a surface model?
     - Atmosphere Priors: How to configure the atmosphere parameters?
     - Aerosol Models: How to choose and configure aerosol options?
- Interpreting ISOFIT Output:
     - Visualizing Output: How to plot and inspect the reflectance values?
     - Uncertainty Analysis: How to inspect and interpret the uncertainties?
     - Parameter Sensitivity Analysis: How to understand the impact of different parameters?
- New Missions:
     - How to run ISOFIT on data from new instruments?
     - Noise Models: How to create the instrument noise models?
     - Instrument Calibration: How to evaluate spectral and radiometric calibrations?