isofit/isofit #511: running example

jenniferbuz, 2024-07-01:
hello! I'm running isofit after a hiatus and just downloaded the latest version. 

In running the example file first the surface model is not specified in the script. That is an easy fix, I pointed it to the surface model in the Pasadena example. However now I have run into a missing package and a netCDF error. 

before installing requests-unixsocket:
```
Test zip already present, skipping download
2024-07-01 19:49:58,945 INFO worker.py:1752 -- Started a local Ray instance.
INFO:2024-07-01,19:49:59 || apply_oe.py:apply_oe() | namespace(aerosol_climatology_path=None, analytical_line=False, atm_sigma=(2.0,), atmosphere_type='ATM_MIDLAT_SUMMER', channelized_uncertainty_path=None, copy_input_files=False, empirical_line=False, emulator_base='/home/ubuntu/share/Team/Jennifer/sRTMnet_v100/', input_loc='small_chunk/ang20170323t202244_loc_7000-7010', input_obs='small_chunk/ang20170323t202244_obs_7000-7010', input_radiance='small_chunk/ang20170323t202244_rdn_7000-7010', log_file=None, logging_level='INFO', lut_config_file=None, memory_gb=-1, model_discrepancy_path=None, modtran_path=None, multiple_restarts=False, n_cores=4, num_cpus=1, num_neighbors=(), prebuilt_lut=None, presolve=True, pressure_elevation=True, ray_temp_dir='/tmp/ray', rdn_factors_path=None, segmentation_size=400, sensor='ang', surface_category='multicomponent_surface', surface_path='/home/ubuntu/isofit/examples/20171108_Pasadena/configs/ang20171108t184227_surface.json', wavelength_path=None, working_directory='small_chunk/L2A_reflectance')
INFO:2024-07-01,19:49:59 || apply_oe.py:apply_oe() | Checking input data files...
INFO:2024-07-01,19:49:59 || apply_oe.py:apply_oe() | ...Data file checks complete
INFO:2024-07-01,19:49:59 || apply_oe.py:apply_oe() | Setting up files and directories....
INFO:2024-07-01,19:49:59 || template_construction.py:__init__() | Flightline ID: ang20170323t202244
INFO:2024-07-01,19:49:59 || template_construction.py:__init__() | no noise path found, proceeding without
INFO:2024-07-01,19:49:59 || template_construction.py:stage_files() | Staging /home/ubuntu/isofit/examples/20171108_Pasadena/configs/ang20171108t184227_surface.json to /home/ubuntu/isofit/examples/image_cube/small_chunk/L2A_reflectance/data/surface.mat
INFO:2024-07-01,19:49:59 || apply_oe.py:apply_oe() | ...file/directory setup complete
INFO:2024-07-01,19:49:59 || apply_oe.py:apply_oe() | No wavelength file provided. Obtaining wavelength grid from ENVI header of radiance cube.
INFO:2024-07-01,19:49:59 || apply_oe.py:apply_oe() | Wavelength units of nm inferred...converting to microns
INFO:2024-07-01,19:49:59 || template_construction.py:check_surface_model() | No surface model provided. Build new one using given config file.
0 ['../../../data/reflectance/surface_model_ucsb']
1 ['../../../data/reflectance/surface_model_ucsb']
2 ['../../../data/reflectance/surface_model_ucsb']
3 ['../../../data/reflectance/surface_model_ucsb']
4 ['../../../data/reflectance/surface_model_ucsb']
5 ['../../../data/reflectance/surface_model_ucsb']
6 ['../../../data/reflectance/surface_model_ucsb']
7 ['../../../data/reflectance/surface_model_ucsb']
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | Observation means:
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | Path (km): 20.716992087408016
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | To-sensor azimuth (deg): 71.27130737245439
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | To-sensor zenith (deg): 162.6547171717885
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | To-sun zenith (deg): 32.456986056768166
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | Relative to-sun azimuth (deg): 53.94754995986687
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | Altitude (km): 19.808562728402368
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | Pre-solve H2O grid: [0.01 0.67 1.34 2.   2.67 3.33 4.   4.66 5.33 5.99]
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | Writing H2O pre-solve configuration file.
INFO:2024-07-01,19:50:03 || apply_oe.py:apply_oe() | Run ISOFIT initial guess
INFO:2024-07-01,19:50:03 || configs.py:create_new_config() | Loading config file: /home/ubuntu/isofit/examples/image_cube/small_chunk/L2A_reflectance/config/ang20170323t202244_h2o.json
INFO:2024-07-01,19:50:03 || configs.py:get_config_errors() | Checking config sections for configuration issues
INFO:2024-07-01,19:50:03 || configs.py:get_config_errors() | Configuration file checks complete, no errors found.
2024-07-01 19:50:03,369 INFO worker.py:1585 -- Calling ray.init() again after it has already been called.
INFO:2024-07-01,19:50:03 || isofit.py:run() | Building first forward model, will generate any necessary LUTs
INFO:2024-07-01,19:50:03 || radiative_transfer_engine.py:__init__() | Loading from wavelength_file: /home/ubuntu/isofit/examples/image_cube/small_chunk/L2A_reflectance/data/wavelengths.txt
INFO:2024-07-01,19:50:03 || radiative_transfer_engine.py:__init__() | No LUT store found, beginning initialization and simulations
INFO:2024-07-01,19:50:03 || radiative_transfer_engine.py:__init__() | Initializing LUT file
Traceback (most recent call last):
  File "/home/ubuntu/.local/bin/isofit", line 8, in <module>
    sys.exit(cli())
  File "/home/ubuntu/.local/lib/python3.8/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/ubuntu/.local/lib/python3.8/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/ubuntu/isofit/isofit/utils/apply_oe.py", line 100, in cli_apply_oe
    apply_oe(SimpleNamespace(**kwargs))
  File "/home/ubuntu/isofit/isofit/utils/apply_oe.py", line 513, in apply_oe
    retrieval_h2o.run()
  File "/home/ubuntu/isofit/isofit/core/isofit.py", line 110, in run
    self.fm = fm = ForwardModel(self.config)
  File "/home/ubuntu/isofit/isofit/core/forward.py", line 82, in __init__
    self.RT = RadiativeTransfer(self.full_config)
  File "/home/ubuntu/isofit/isofit/radiative_transfer/radiative_transfer.py", line 101, in __init__
    rte = RTE[confRT.engine_name](**params)
  File "/home/ubuntu/isofit/isofit/radiative_transfer/radiative_transfer_engine.py", line 224, in __init__
    self.lut = luts.Create(
[... 214 lines cut ...]
    datasets = [open_(p, **open_kwargs) for p in paths]
  File "/home/ubuntu/.local/lib/python3.8/site-packages/xarray/backends/api.py", line 541, in open_dataset
    backend_ds = backend.open_dataset(
  File "/home/ubuntu/.local/lib/python3.8/site-packages/xarray/backends/netCDF4_.py", line 578, in open_dataset
    store = NetCDF4DataStore.open(
  File "/home/ubuntu/.local/lib/python3.8/site-packages/xarray/backends/netCDF4_.py", line 382, in open
    return cls(manager, group=group, mode=mode, lock=lock, autoclose=autoclose)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/xarray/backends/netCDF4_.py", line 329, in __init__
    self.format = self.ds.data_model
  File "/home/ubuntu/.local/lib/python3.8/site-packages/xarray/backends/netCDF4_.py", line 391, in ds
    return self._acquire()
  File "/home/ubuntu/.local/lib/python3.8/site-packages/xarray/backends/netCDF4_.py", line 385, in _acquire
    with self._manager.acquire_context(needs_lock) as root:
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/xarray/backends/file_manager.py", line 197, in acquire_context
    file, cached = self._acquire_with_cache_info(needs_lock)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/xarray/backends/file_manager.py", line 215, in _acquire_with_cache_info
    file = self._opener(*self._args, **kwargs)
  File "src/netCDF4/_netCDF4.pyx", line 2470, in netCDF4._netCDF4.Dataset.__init__
  File "src/netCDF4/_netCDF4.pyx", line 2107, in netCDF4._netCDF4._ensure_nc_success
OSError: [Errno -101] NetCDF: HDF error: '/home/ubuntu/isofit/examples/image_cube/small_chunk/L2A_reflectance/lut_h2o/lut.nc'
```

I see there is a 'ModuleNotFoundError: No module named 'apt'' in there but I have no idea what package this is by that name

jammont, 2024-09-09:
This looks to be an issue with your Python environment and not explicitly ISOFIT related. Can you try creating a new environment and testing again? If the issue persists, please post the steps you took to create an environment and install ISOFIT.

jenniferbuz, 2024-11-04:
I was able to get it running with a new environment so I will close this

closed 2024-11-04 by jenniferbuz