isofit/isofit #583: Improve 6S input/output validation

danschef, 2024-11-08:
I just ran some tests on EnMAP data together with @unbohn and we had the following error in the log:

```python
(streamSimulation pid=1041734) Failed to parse any data for file: /path/to/workdir/lut_full/AOT550-1.0000_H2OSTR-0.8905
(streamSimulation pid=1041734) No data was returned for point [1.     0.8905]
```
Also the surface reflectance coming out of isofit did not look that good:

[image]

After some investigation, it turned out that all the AOT files at `/path/to/workdir/lut_full/` produced by 6S were empty. We ran one of the `/path/to/workdir/lut_full/LUT_AOT550*.sh` files manually and 6S indeed produced this error:

```fortran
At line 1043 of file main.f (unit = 5, file = 'stdin')
Fortran runtime error: Bad real number in item 1 of list input

Error termination. Backtrace:
#0  0x7f1a2e458960 in ???
#1  0x7f1a2e4594d9 in ???
#2  0x7f1a2e45a10f in ???
#3  0x7f1a2e6a51b6 in ???
#4  0x7f1a2e6a65fd in ???
#5  0x7f1a2e6a72aa in ???
#6  0x563670a4db1b in ???
#7  0x563670a5d7c7 in ???
#8  0x7f1a2e14ed8f in __libc_start_call_main
        at ../sysdeps/nptl/libc_start_call_main.h:58
#9  0x7f1a2e14ee3f in __libc_start_main_impl
        at ../csu/libc-start.c:392
#10  0x563670a4c354 in ???
#11  0xffffffffffffffff in ???
```

So we looked into the input file of the `LUT_AOT550*.sh` (`LUT_AOT550-*.inp`) and it seems to be caused by one of the illumination angles (likely the solar zenith angle above 70 degrees):

```bash
0 (User defined)
71.13128265634927 -106.38034756655972 29.86408497768096 12.858726946350673 7 11
8  (User defined H2O, O3)
1.2787, 0.3
1
0
1.0
-1.51 (target level)
--562.18 (sensor level)
-1.2787, -0.3
1.0
-2
0.35
2.5
0 Homogeneous surface
0 (no directional effects)
0
0
0
-1 No atm. corrections selected
```

However, the `Fortran runtime error: Bad real number in item 1 of list input` was not propagated to the isofit output, so it was a bit hard to find out what went wrong.

Therefore, I suggest to add a check into isofit which validates the content of the input files of 6S or at least raises a useful error message if all the 6S outputs (the AOT files) are empty.

danschef, 2024-11-08:
@unbohn

unbohn, 2024-11-09:
@danschef Thanks for opening this issue! Looking at the 6s .inp file again, I think the failure of the runs could have also been caused by the double minus sign in front of the sensor level (`--562.18`). It should actually be `-562.18`. Maybe you could double-check where that additional `-` comes from. That doesn't change the purpose of this issue though as we would still need a useful error message in case the 6s runs were not successful.

danschef, 2024-11-27:
I fixed the `--562.18` for the sensor altitude [here](https://git.gfz-potsdam.de/EnMAP/GFZ_Tools_EnMAP_BOX/EnPT/-/merge_requests/106/diffs?commit_id=4855bfb20c5a58d16deac1272b51395c7c30ee9d). On the ISOFIT side, this could easily be avoided by checking if the value of `altitude_km` is positive in [write_modtran_template()](https://github.com/isofit/isofit/blob/a38fce58b2942f97d8863b6123562fc1261124e8/isofit/utils/template_construction.py#L1020). This would also be meaningful because the `SIXS_TEMPLATE` [adds a minus to the altitude](https://github.com/isofit/isofit/blob/a38fce58b2942f97d8863b6123562fc1261124e8/isofit/radiative_transfer/engines/six_s.py#L47) without checking the actual value of the incoming altitude.

After fixing the double minus, 6s indeed runs through. I am just wondering why all `*.inp` files now contain `-99.00 (sensor level)`. However, [this line](https://github.com/isofit/isofit/blob/a38fce58b2942f97d8863b6123562fc1261124e8/isofit/radiative_transfer/engines/six_s.py#L226) looks like this is intentional.

Now I still get some warnings when running ISOFIT but the output spectra are not flat anymore:

[image]

unbohn, 2024-11-28:
Perfect, looks way better now! I'll add a check for the value of `altitude_km`. To explain the uniform sensor level in all input files, 6s can only simulate a max. sensor altitude of 99 km, thus, we restrict this parameter even if the actual observer altitude is higher.

pgbrodrick, 2024-11-28:
Definitely looking reasonable.  We could add a check to altitude - [we have one](https://github.com/isofit/isofit/blob/a38fce58b2942f97d8863b6123562fc1261124e8/isofit/utils/apply_oe.py#L395-L399) for elevation, because that can reasonably go negative and 6S can't handle this behavior...we could add a warning for negative altitudes.  It does get hard to safeguard all geometry values, but that one's simple enough.

unbohn, 2024-11-30:
@pgbrodrick I re-open this issue as I'm still planning to implement a check that validates the content of the 6s output files. In particular and following @danschef's suggestion, we should make sure to raise a useful error message if the files are empty because of internal 6s Fortran runtime errors. #599 did not cover this and I accidentally proposed to close #583.