isofit/isofit #601: Scientific notation floats in isofit_config_default.json incorrectly parsed as strings

danschef, 2024-12-02:
I just noted that using float values in the scientific notation like `1e-10` in the ISOFIT configuration file (I use a JSON file) leads to a TypeError like this:

```python
test_isofit_enmap.py:122: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../enpt/processors/atmospheric_correction/_isofit_enmap.py:566: in _run
    Isofit(
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/core/isofit.py:71: in __init__
    self.config = configs.create_new_config(config_file)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/configs/configs.py:174: in create_new_config
    config = Config(config_dict)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/configs/configs.py:88: in __init__
    self.set_config_options(configdict)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/configs/base_config.py:45: in set_config_options
    sub_config = keytype(configdict[key])
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/configs/sections/forward_model_config.py:58: in __init__
    self.set_config_options(sub_configdic)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/configs/base_config.py:45: in set_config_options
    sub_config = keytype(configdict[key])
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <isofit.configs.sections.radiative_transfer_config.RadiativeTransferConfig object at 0x7facc22b39b0>
sub_configdic = {'lut_grid': {'AOT550': [0.001, 0.1009, 0.2008, 0.3007, 0.4006, 0.5005, ...], 'H2OSTR': ['1e-10', 1.0, 1.5, 2.0, 2.7, ...: {'bounds': [0.001, 5.0], 'init': 2.5, 'prior_mean': 2.5, 'prior_sigma': 100.0, ...}}, 'unknowns': {'H2O_ABSCO': 0.0}}

    def __init__(self, sub_configdic: dict = None):
        self._statevector_type = StateVectorConfig
        self.statevector: StateVectorConfig = StateVectorConfig({})
    
        self._lut_grid_type = OrderedDict
        self.lut_grid = None
    
        self._unknowns_type = RadiativeTransferUnknownsConfig
        self.unknowns: RadiativeTransferUnknownsConfig = None
    
        self._interpolator_style_type = str
        self.interpolator_style = "mlg"
        """str: Style of interpolation.
        - mlg   = Multilinear Grid
        - rg    = RegularGrid
        Speed performance:
            mlg >> stacked rg >> unstacked rg
        Caching provides significant gains for rg, marginal for mlg"""
    
        self._overwrite_interpolator_type = bool
        self.overwrite_interpolator = False
        """bool: Overwrite any existing interpolator pickles"""
    
        self._cache_size_type = int
        self.cache_size = 16
        """int: Size of the cache to store interpolation lookups. Defaults to 16 which
        provides the most significant gains. Setting higher may provide marginal gains."""
    
        self.set_config_options(sub_configdic)
    
        # sort lut_grid
        for key, value in self.lut_grid.items():
>           self.lut_grid[key] = sorted(self.lut_grid[key])
E           TypeError: '<' not supported between instances of 'float' and 'str'

../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/configs/sections/radiative_transfer_config.py:331: TypeError
```

This is because the 1e-10 is parsed as string instead of a float:

[image]

Here is a [related discussion](https://stackoverflow.com/questions/30458977/yaml-loads-5e-6-as-string-and-not-a-number) and a [possible fix](https://stackoverflow.com/questions/30458977/yaml-loads-5e-6-as-string-and-not-a-number).

evan-greenbrg, 2024-12-04:
It's interesting that the this is caused by the yaml regex behavior. Based on the Stackoverflow discussion and the [yaml docs](https://yaml.org/spec/1.2-old/spec.html#id2804092) it points to, the regex for parsing the yaml file will interpret a float led number in scientific notation (e.g. 1.0e-6), but not an int-led number (e.g. 1e-6).

You can actually see this behavior from the regex: `/-?[1-9]\.[0-9]*[1-9]e[-+][1-9][0-9]*/` in something like regext101.com, where it matches `1.0e-6`, but not `1e-6`.

Either way it's worth a discussion whether we should add an additional parser/check when we load the lut, simply convert lut grid values to float (`float('1e-6')` does work), or handle the incorrect config value with a correctly caught error.

pgbrodrick, 2025-04-07:
Mostly concerning if this happens parsing surface files as well - to investigate.