isofit/isofit #602: NameError: name 'Logger' is not defined

danschef, 2024-12-03:
I just ran into an error due to an undefined Logger variable [here](https://github.com/isofit/isofit/blob/70fc60f3f8514529c8812f52326dbad5d6c6e838/isofit/radiative_transfer/radiative_transfer.py#L101):

```python
test_isofit_enmap.py:122: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../enpt/processors/atmospheric_correction/_isofit_enmap.py:571: in _run
    ).run(row_column=None)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/core/isofit.py:114: in run
    self.fm = fm = ForwardModel(self.config)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/core/forward.py:74: in __init__
    self.RT = RadiativeTransfer(self.full_config)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/radiative_transfer/radiative_transfer.py:101: NameError

NameError: name 'Logger' is not defined
```

Should be an easy fix..

evan-greenbrg, 2024-12-03:
Thanks for catching this! It looks like the Logger definition is missing from radiative_transfer.py. I agree that this should be an easy fix. I'll put a PR together. 

The missing Logger is catching because the config is failing a built-in error check between the length of the config statevectors and the and the length of the rte statevector elements. 
```python
            # Make sure the length of the config statevectores match the engine's assumed statevectors
            if (expected := len(config.statevector.get_element_names())) != (
                got := len(rte.indices.x_RT)
            ):
                error = f"Mismatch between the number of elements for the config statevector and LUT.indices.x_RT: {expected=}, {got=}"
                Logger.error(error)
                raise AttributeError(error)
```

You may have already resolved this, but in the meantime, I can also help resolve the config mismatch, which would circumvent the Logger issue.

evan-greenbrg, 2024-12-03:
It's hard to know exactly what your config set-up error is without looking at your config directly, but my guess is that it is along these lines:

I was able to reproduce the error by passing `isofit run` a config where in the config JSON the length of the `statevector_names` entry does not match the number of keys in the `statevector` entry.

The root of the NameError was spot on and adding the missing definition fixed that issue. Just wanted to give some context around how to fix the AttributeError that will pop up once the Logger is fixed before this issue is closed.

danschef, 2024-12-04:
Thanks @evan-greenbrg, the reason for running into this error was in my particular case that I provided a LUT which had a variable and dimension called "surface_elevation". However, ISOFIT  expects it be named "surface_elevation_km". Therefore, surface_elevation was not recognized as a geometry variable which caused the mismatch.

To me, the error message itself makes sense from the developer point of view but requires to know what `LUT.indices.x_RT` actually is (the usual user does not know this). Yesterday, @unbohn sent me the naming convention of the variables and dimensions that should be contained in the LUT (at least for the geometry variables) and I am wondering why ISOFIT not simply performs a check if the actual LUT variables conform to this naming convention:

```python
geometry_input_names = [
        "observer_azimuth",
        "observer_zenith",
        "solar_azimuth",
        "solar_zenith",
        "relative_azimuth",
        "observer_altitude_km",
        "surface_elevation_km",
    ]
```

This should avoid subsequent issues as the one above and should provide the user a less cryptic error message.

pgbrodrick, 2024-12-05:
First comment (logging import) now handled, leaving open for addressing the other comments.

PR #604 (2024-12-05): Added Logger definition to radiative_transfer.py
Will close #602