isofit/isofit #603: examples not running

gswhitti, 2024-12-03:
I'm trying to use ISOFIT for the first time and the python and shell scripts in examples don't seem to run.  My short-term goal is to be able to use ISOFIT to calibrate an AVIRIS or AVIRIS-NG cube to reflectance as a step towards calibrating data from another VNIR/SWIM sensor.  What's the best way to get started?  I'm not having success following the docs as almost all of the examples I've tried fail.

Following https://registry.hub.docker.com/r/jammont/isofit

```
E:\isofit_test>docker run --rm -it --shm-size=16gb jammont/isofit:3.2.2 bash examples/image_cube/small/analytical.sh
2024-12-03 14:29:01,178 INFO worker.py:1819 -- Started a local Ray instance.
INFO:2024-12-03,14:29:01 || apply_oe.py:apply_oe() | namespace(surface_path='/home/mambauser/examples/image_cube/small/configs/surface.json', emulator_base='/home/mambauser/srtmnet/sRTMnet_v120.h5', n_cores=4, presolve=True, segmentation_size=400, pressure_elevation=True, analytical_line=True, input_radiance='/home/mambauser/imagecube/small/ang20170323t202244_rdn_7000-7010', input_loc='/home/mambauser/imagecube/small/ang20170323t202244_obs_7000-7010', input_obs='/home/mambauser/imagecube/small/ang20170323t202244_loc_7000-7010', working_directory='/home/mambauser/examples/image_cube/small', sensor='ang', copy_input_files=False, modtran_path=None, wavelength_path=None, surface_category='multicomponent_surface', aerosol_climatology_path=None, rdn_factors_path=None, atmosphere_type='ATM_MIDLAT_SUMMER', channelized_uncertainty_path=None, model_discrepancy_path=None, lut_config_file=None, multiple_restarts=False, logging_level='INFO', log_file=None, num_cpus=1, memory_gb=-1, empirical_line=False, ray_temp_dir='/tmp/ray', num_neighbors=(), atm_sigma=(2.0,), prebuilt_lut=None, inversion_windows=(), no_min_lut_spacing=False)
INFO:2024-12-03,14:29:01 || apply_oe.py:apply_oe() | Checking input data files...
INFO:2024-12-03,14:29:01 || apply_oe.py:apply_oe() | ...Data file checks complete
INFO:2024-12-03,14:29:01 || apply_oe.py:apply_oe() | Setting up files and directories....
INFO:2024-12-03,14:29:01 || template_construction.py:__init__() | Flightline ID: ang20170323t202244
INFO:2024-12-03,14:29:01 || template_construction.py:__init__() | no noise path found, proceeding without
INFO:2024-12-03,14:29:01 || template_construction.py:stage_files() | Staging /home/mambauser/examples/image_cube/small/configs/surface.json to /home/mambauser/examples/image_cube/small/data/surface.mat
INFO:2024-12-03,14:29:01 || apply_oe.py:apply_oe() | ...file/directory setup complete
INFO:2024-12-03,14:29:01 || apply_oe.py:apply_oe() | Using inversion windows: [[350.0, 1360.0], [1410, 1800.0], [1970.0, 2500.0]]
Traceback (most recent call last):
  File "/opt/conda/envs/isofit/bin/isofit", line 8, in <module>
    sys.exit(cli())
  File "/opt/conda/envs/isofit/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/opt/conda/envs/isofit/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/opt/conda/envs/isofit/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/opt/conda/envs/isofit/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/opt/conda/envs/isofit/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/mambauser/isofit/isofit/utils/apply_oe.py", line 102, in cli_apply_oe
    apply_oe(SimpleNamespace(**kwargs))
  File "/home/mambauser/isofit/isofit/utils/apply_oe.py", line 312, in apply_oe
    ) = tmpl.get_metadata_from_obs(paths.obs_working_path, lut_params)
  File "/home/mambauser/isofit/isofit/utils/template_construction.py", line 1457, in get_metadata_from_obs
    to_sun_azimuth = obs[:, :, 3]
  File "/opt/conda/envs/isofit/lib/python3.10/site-packages/numpy/core/memmap.py", line 335, in __getitem__
    res = super().__getitem__(index)
IndexError: index 3 is out of bounds for axis 2 with size 3
```

Running python scripts interactively in docker bash shell (so conda environment is activated).

```
E:\isofit_test>docker run --rm -it --shm-size=16gb jammont/isofit:3.2.2 bash
(isofit) mambauser@9dbb967b0d08:~$
(isofit) mambauser@9dbb967b0d08:~$ cd examples/scripts/
(isofit) mambauser@09e3524786fc:~/examples/scripts$ ls
test_6S.py  test_modtran.py  test_sRTMnet.py
(isofit) mambauser@09e3524786fc:~/examples/scripts$ python test_6S.py
Traceback (most recent call last):
  File "/home/mambauser/examples/scripts/test_6S.py", line 8, in <module>
    from isofit.radiative_transfer.six_s import SixSRT
ModuleNotFoundError: No module named 'isofit.radiative_transfer.six_s'
(isofit) mambauser@9dbb967b0d08:~/examples/scripts$ python test_sRTMnet.py
Traceback (most recent call last):
  File "/home/mambauser/examples/scripts/test_sRTMnet.py", line 7, in <module>
    from isofit.radiative_transfer.sRTMnet import SimulatedModtranRT
ModuleNotFoundError: No module named 'isofit.radiative_transfer.sRTMnet'
```
I've also had similar issues trying to use the jupyter notebooks in isotutorials.

jammont, 2024-12-10:
@unbohn the `IndexError` is fixed by https://github.com/isofit/isofit-tutorials/pull/12, right?

As for the second issue with executing the scripts directly in the container, those files under the `scripts/` directory are:
1. out of date
2. intended to be used by developers and not as an example for users

We'll seek to update these and add some documentation to indicate the above intentions. For now, I recommend leaning on the SantaMonic and Pasadena examples as those should both be fully working.

unbohn, 2024-12-10:
@jammont Yes, that's correct.

pgbrodrick, 2025-04-07:
Resolved with latest release!

closed 2025-04-07 by pgbrodrick