isofit/isofit #605: TypeError in case user-provided LUT does not contain "coszen" variable

danschef, 2024-12-04:
In case a user-provided LUT does not contain a `coszen` variable, ISOFIT raises a `TypeError` [here](https://github.com/isofit/isofit/blob/70fc60f3f8514529c8812f52326dbad5d6c6e838/isofit/inversion/inverse_simple.py#L119) because the [RadiativeTransfer.coszen](https://github.com/isofit/isofit/blob/70fc60f3f8514529c8812f52326dbad5d6c6e838/isofit/radiative_transfer/radiative_transfer.py#L158) returns `None`:

```python
test_isofit_enmap.py:122: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../enpt/processors/atmospheric_correction/_isofit_enmap.py:573: in _run
    ).run(row_column=None)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/core/isofit.py:174: in run
    self.workers.map_unordered(
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/debug/ray_bypass.py:105: in map_unordered
    return [func(self.actors, item).get() for item in iterable]
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/debug/ray_bypass.py:39: in get
    return self.obj(*self.args, **self.kwargs)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/core/isofit.py:242: in run_set_of_spectra
    states = self.iv.invert(input_data.meas, input_data.geom)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/inversion/inverse.py:337: in invert
    x0 = invert_simple(self.fm, meas, geom)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/inversion/inverse_simple.py:447: in invert_simple
    x[forward.idx_RT] = heuristic_atmosphere(
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/inversion/inverse_simple.py:119: TypeError

TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'
```

This should somehow be handled and a more useful error message should be raised.

unbohn, 2024-12-06:
Thanks for raising this issue, @danschef! As you mention in #606, we plan to replace the static `coszen` from the LUT by the per-pixel value obtained from the `Geometry` object sometime soon. While that will prevent us from running into the above-mentioned issue, a check of the LUT-based `coszen` and a useful error message would of course be a needed improvement.

danschef, 2025-01-21:
I just noticed that there is also a similar error in case I provide my pre-built LUT in rdn mode but without a "coszen" variable in the NetCDF file:

```python
test_isofit_enmap.py:237: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../enpt/processors/atmospheric_correction/_isofit_enmap.py:611: in _run
    ).run(row_column=None)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/core/isofit.py:174: in run
    self.workers.map_unordered(
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/debug/ray_bypass.py:105: in map_unordered
    return [func(self.actors, item).get() for item in iterable]
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/debug/ray_bypass.py:39: in get
    return self.obj(*self.args, **self.kwargs)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/core/isofit.py:247: in run_set_of_spectra
    self.io.write_spectrum(row, col, states, self.fm, self.iv)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/core/fileio.py:741: in write_spectrum
    to_write = self.build_output(states, self.current_input_data, fm, iv)
../../../../miniforge3/envs/enpt_dev_isofit321/lib/python3.12/site-packages/isofit/core/fileio.py:697: TypeError

TypeError: unsupported operand type(s) for *: 'float' and 'NoneType'
```

unbohn, 2025-01-21:
Ah, it looks like you're hitting a very edge case here. In case the config file includes the key `atmospheric_coefficients_file` as part of the output datasets, ISOFIT attempts to write the interpolated LUT quantities (path radiance, spherical albedo, etc.) from each inversion to a file. We haven't actually updated this section (as it's barely used), so that it still assumes the `transm` mode by default and thus, tries to multiply path reflectance and transmittance by `coszen`. A quick fix for you would be to remove `atmospheric_coefficients_file` from the config, but we will of course work on fixing this issue anyhow.