isofit/isofit #649: Error on sixs download

brobertspierel, 2025-03-10:
Apologies if this has come up in my absence but using the new (to me) download routines with ISOFIT I am able to get SRTMnet without any issue but when I do `isofit download sixs` following the installation instructions I get a NameError: `NameError: name 'validate_' is not defined`. I can reassign the path to a previously build version of 6S or re-install following the advanced instructions as I did in the past but thought I would flag this issue as it seems like great functionality to have access to.

jammont, 2025-03-11:
This issue is being fixed in #648. Thanks for the ticket!

brobertspierel, 2025-03-11:
Ah perfect, I quickly looked to see if there was another issue but I guess I missed #648. Thanks!

PR #648 (2025-03-13): Downloaders Bugfixes
Fixed a few issues with the downloader modules when working with a user to fix their installation:
- `isofit download sixs` was broken due to having the wrong parameters
- Added more context to one of the validation errors
- Disabled piping the error output of the `make` call, hopefully future errors will be outputted rather than quietly being discarded
- Fixed `isofit download imagecube` broken when using a non-default path
- Fixed the example building from templates retrieving non-json files
  - MacOS Finder likes to create `.DS_Store` in every directory to casually poll [... cut]