isofit/isofit #658: Coupling terms not found for the SimulatedModtranRT

samuelleblanc, 2025-03-26:
While trying to run isofit for hyperion, recently updated to v3.4.1 I get this error:

> 2025-03-26 19:04:16,669	ERROR worker.py:422 -- Unhandled error (suppress with 'RAY_IGNORE_UNHANDLED_ERRORS=1'): ray::Worker.run_set_of_spectra() (pid=2191831, ip=192.168.10.201, actor_id=67f27d0a41dd5e75ee3b05de01000000, repr=<isofit.core.isofit.Worker object at 0x7f24602ad2b0>)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/sam/mambaforge/envs/isofit_env/lib/python3.12/site-packages/isofit/core/isofit.py", line 242, in run_set_of_spectra
>     states = self.iv.invert(input_data.meas, input_data.geom)
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/sam/mambaforge/envs/isofit_env/lib/python3.12/site-packages/isofit/inversion/inverse.py", line 379, in invert
>     Seps_inv, Seps_inv_sqrt = self.calc_Seps(x, meas, geom)
>                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/sam/mambaforge/envs/isofit_env/lib/python3.12/site-packages/isofit/inversion/inverse.py", line 217, in calc_Seps
>     Seps = self.fm.Seps(x, meas, geom)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/sam/mambaforge/envs/isofit_env/lib/python3.12/site-packages/isofit/core/forward.py", line 245, in Seps
>     Kb = self.Kb(x, geom)
>          ^^^^^^^^^^^^^^^^
>   File "/home/sam/mambaforge/envs/isofit_env/lib/python3.12/site-packages/isofit/core/forward.py", line 321, in Kb
>     rdn_hi = self.calc_rdn(x, geom, rho_dir_dir, rho_dif_dir, Ls=Ls)
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/sam/mambaforge/envs/isofit_env/lib/python3.12/site-packages/isofit/core/forward.py", line 209, in calc_rdn
>     return self.RT.calc_rdn(x_RT, rho_dir_dir_hi, rho_dif_dir_hi, Ls_hi, geom)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/sam/mambaforge/envs/isofit_env/lib/python3.12/site-packages/isofit/radiative_transfer/radiative_transfer.py", line 205, in calc_rdn
>     L_dir_dir, L_dif_dir, L_dir_dif, L_dif_dif = self.get_L_coupled(
>                                                  ^^^^^^^^^^^^^^^^^^^
>   File "/home/sam/mambaforge/envs/isofit_env/lib/python3.12/site-packages/isofit/radiative_transfer/radiative_transfer.py", line 388, in get_L_coupled
>     for key in self.rt_engines[0].coupling_terms
>                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> AttributeError: 'SimulatedModtranRT' object has no attribute 'coupling_terms'

[isofit_v3_4_1_hyperion_extra_outputs.txt](https://github.com/user-attachments/files/19474596/isofit_v3_4_1_hyperion_extra_outputs.txt)

evan-greenbrg, 2025-03-26:
Hey Sam, happy to try and debug. Would you mind sharing isofit config for this run?  It looks like it should follow something like the format `EO1H0400332013093110K7_isofit.json` and is found within the `config/` subdirectory.

samuelleblanc, 2025-03-26:
Here is that file.

[EO1H0400332013093110K7_isofit.json](https://github.com/user-attachments/files/19475821/EO1H0400332013093110K7_isofit.json)

evan-greenbrg, 2025-03-26:
Thanks! So two things jump out at me at being strange. The first is that the `multipart_transmittance` flag is `true`. This flag tells the routine which radiance model to use as the forward model. The case where `multipart_transmittance = true` is an area of active development. 

The second thing is that the config says the engine is modtran, but the error calls the `SimulatedModtranRT` object, which is the sRTMnet emulator. I'm curious how this happened.

First thing I'm going to double check is if using modtran as the engine in applyOE defaults to `multipart_transmittance = true`. As a quick clarifying question, is Modtran your preferred radiative transfer model, or would you use either Modtran or sRTMnet?

samuelleblanc, 2025-03-26:
I'm specyfying sRTMnet path (which is what I want) in the python apply_oe, and keeping modtran_path to None. I'm not sure about the multipart_transmittance flag.

evan-greenbrg, 2025-03-26:
That's great to know. My first guess to what is happening is that something is incorrect on the initial ApplyOE call. It sounds like the config you sent over does not reflect what you want to run (SRTMnet and multipart_transmittance = False). I see some of the inputs on the file you shared. 

Is it easy for you to send over the ApplyOE call you are making?

samuelleblanc, 2025-03-28:
That call was the issue. I was using the 'SimpleNamespace' class for the call, which now does not seem to work properly. 

>args = SimpleNamespace(**{
>    'input_radiance'   : paths['radiance'][:-4], # Radiance
>    'input_loc'         : paths['loc'][:-4], # Location
>    'input_obs'         : paths['obs'][:-4], # Observations
>    'working_directory' : paths['working_directory'], # Output directory
>    'ray_temp_dir': "/tmp/ray",
>    'logging_level': "INFO",
>    'log_file': None,
>    'sensor'            : "hyp", 
>...
>
> apply_oe.apply_oe(args)

And now after updating to 3.4.1 I need to use a dict instead:

>args = {
>    'input_radiance'   : paths['radiance'][:-4], # Radiance
>    'input_loc'         : paths['loc'][:-4], # Location
>    'input_obs'         : paths['obs'][:-4], # Observations
>    'working_directory' : paths['working_directory'], # Output directory
>    'ray_temp_dir': "/tmp/ray",
>    'logging_level': "INFO",
>...
>
> apply_oe.apply_oe(**args)

This works now.

closed 2025-03-28 by samuelleblanc