isofit/isofit #660: confusing notation in class RadiativeTransfer() in get_L_atm()

brentwilder, 2025-03-28:
Currently if `rt_mode=="rdn"` the data it pulls is still called `rhoatm`. This is likely the correct value, but is a little confusing for someone to come across a rho here. Is it possible this term could be a more generalizable, for example it could be called `l0` or `atmpath`?

https://github.com/isofit/isofit/blob/dev/isofit/radiative_transfer/radiative_transfer.py#L318

```python
   def get_L_atm(self, x_RT: np.array, geom: Geometry) -> np.array:
        """Get the interpolated modeled atmospheric path radiance.

        Args:
            x_RT: radiative-transfer portion of the statevector
            geom: local geometry conditions for lookup

        Returns:
            interpolated modeled atmospheric path radiance
        """
        L_atms = []

        coszen, cos_i = self.check_coszen_and_cos_i(geom)

        for RT in self.rt_engines:
            if RT.treat_as_emissive:
                r = RT.get(x_RT, geom)
                rdn = r["thermal_upwelling"]
                L_atms.append(rdn)
            else:
                r = RT.get(x_RT, geom)
                if RT.rt_mode == "rdn":
                    L_atm = r["rhoatm"]
                else:
                    rho_atm = r["rhoatm"]
                    L_atm = self.rho_to_rdn(rho_atm, coszen)
                L_atms.append(L_atm)
        return np.hstack(L_atms)

```

davidraythompson, 2025-03-28:
Yes, agree!

brentwilder, 2026-08-25:
Revisiting this a year later, and I think that while renaming keys is not a great idea, we could consider adding to the docs page of what each key represents and what is needed essentially to create a prebuilt-LUT that conforms to how ISOFIT will expect it. 

At a very low level, it could be somewhere on the docs (e.g., https://isofit.github.io/isofit/latest/getting_started/quickstarts/modtran/) that defines each of the terms that can be populated in an atmospheric RT LUT:

- `wl` - wavelength grid in nanometers
- `rhoatm` - path reflectance (or path radiance depending on mode)
- `salb` - atmospheric spherical albedo
- `transm_down_dir` - downward direct transmittance
- `transm_down_dif`- downward diffuse transmittance
- `transm_up_dir` - upward direct transmittance
- `transm_up_dif`- upward diffuse transmittance
- `dir-dir` - coupled term in either radiance or transmittance space (direct x direct)
- `dir-dif` -  "" ""  (direct x diffuse)
- `dif-dir`-  "" ""  (diffuse x direct)
- `dif-dif`-  "" ""  (diffuse x diffuse)

A really nice solution would be if a way we could do this in a semi-automated that pulls straight from https://github.com/isofit/isofit/blob/dev/isofit/atmosphere/atmosphere.py which already contains a lot of information on this.