isofit/isofit #661: MODTRAN day of year irradiance adjustment

pgbrodrick, 2025-03-30:
At present, sRTMnet includes an earth-sun distance adjustment based on the day of the year.  MODTRAN will internally consider this, but I do not believe the current overall workflow (via apply_oe) appropriately sets the template day of year flag for this to get populated.  Furthermore, .nc files built from any of the engines do not pursue an additional day-of-year adjustment.  I propose we:

- remove the sRTMnet adjustment
- store a day-of-year value in the lut netcdf
- track day-of-year in geometry (?) - best place I can think of right now...
- implement a correction to the stored irradiance on load of the .nc

There are other ways to get this done, but we should include in some manner...outside of the bookkeeping, this is essentially 'free' additional accuracy we can gain.

brentwilder, 2025-04-21:
> Do we want to build in support s.t. we can generate a Modtran LUTs using one dayofyear, and making it possible to use them for a different dayofyear?

If the use case was a generalized, massive MODTRAN LUT, perhaps it could be useful? But it would also require that `IDAY`  parameter was set to 93 during generation of LUT. Where in the docs they have, "_IDAY = 93 as the day of year, the average Earth-to-Sun distance_."

evan-greenbrg, 2026-08-31:
Resolved with #1001

closed 2026-08-31 by evan-greenbrg