isofit/isofit #677: isofit SIXS ini documentation

pgbrodrick, 2025-04-29:
Reported by @reginaeckert , isofit gives this advice:

```
ERROR:2025-04-29,10:58:45 || six_s.py:_init_() | WARNING: The environment variable $SIXS_DIR does not match the ISOFIT ini
ERROR:2025-04-29,10:58:45 || six_s.py:_init_() | ENV: /store/shared/sixs
ERROR:2025-04-29,10:58:45 || six_s.py:_init_() | INI: /home/reckert/.isofit/sixs
ERROR:2025-04-29,10:58:45 || six_s.py:_init_() | This may cause issues, please either set the env to the ini, or override the ini to the env using:
ERROR:2025-04-29,10:58:45 || six_s.py:_init_() |   isofit --sixs $SIXS_DIR ...
```

When in fact SIXS is succesfully built in $SIXS_DIR but no in the ini sixs location.  Recommendation to either:

- change the isofit --sixs call to reference the .isofit/sixs directory
- change the recommendation to edit the .ini file to point to SIXS_DIR

@jammont  - recommendation on which way to go?

jammont, 2025-04-29:
> change the recommendation to edit the .ini file to point to SIXS_DIR

That's exactly what the warning is saying. Using `--sixs $SIXS_DIR` used to edit the ini with the env var. This needs to be updated as the syntax has changed to `--path sixs [value]`. By default, this will write back to the ini unless `--no-save` is given. 

Would it be more useful to use our sixs validator to attempt to inform the user which directory is a valid installation? Additionally, do you think there's a better way to describe the above than the way it is written? Seems like it may have caused some confusion.

Side note, we should also change it from `log.error` to `log.warning`.

reginaeckert, 2025-04-29:
Here is the exact output I got from this:
`ERROR:2025-04-28,13:26:05 || six_s.py:__init__() | WARNING: The environment variable $SIXS_DIR does not match the ISOFIT ini
ERROR:2025-04-28,13:26:05 || six_s.py:__init__() | ENV: /store/shared/sixs
ERROR:2025-04-28,13:26:05 || six_s.py:__init__() | INI: /home/reckert/.isofit/sixs
ERROR:2025-04-28,13:26:05 || six_s.py:__init__() | This may cause issues, please either set the env to the ini, or override the ini to the env using:
ERROR:2025-04-28,13:26:05 || six_s.py:__init__() |   isofit --sixs $SIXS_DIR ...`

But when I tried running 
`isofit --sixs \path\to\sixs\dir\ `

or as referenced here https://isofit.readthedocs.io/en/latest/custom/installation.html#quick-start-with-srtmnet-recommended-for-new-users in Manual (Advanced)
`isofit --sixs \path\to\sixs\dir\ build`

I get the error
`Error: No such option: --sixs`

So just noting that all of this is unclear and also if you take it directly (I'm not sure where else to run the command `isofit --sixs \path\to\sixs\dir\ build` than in the terminal?) it throws an error. I ended up editing isofit.ini directly with a text editor to clear this. It didn't make a difference though - the error says, "this may cause issues", and I'm not sure exactly what issues it's referring to. Perhaps if the two versions are different? Or just if one of the installations is broken in some way?

In any case, agreed that it is unclear currently.

pgbrodrick, 2025-05-01:
After discussion, seems like we just need a small update to the cli docs for the newer syntax.

PR #682 (2025-05-08): CLI Docs Update
- Updated docs for the newer ini path overrides syntax
- Added further clarity to the sixs warning when the `$SIXS_DIR` does not match the loaded INI

Closes #677
closed 2025-05-08 by unbohn