isofit/isofit #685: Make pygrib an optional dep

alexjohnson-kobold, 2025-05-09:
`pygrib` is used in only one place, the `HRRR_to_modtran` CLI util. Can we make it an optional dependency? This is coming up now because it has blocked my builds due to the latest Cython release https://github.com/jswhit/pygrib/issues/265 but even without that it would be nice to slim down the install.

I'm happy to make a PR for this if the maintainers agree it would be a positive change.

pgbrodrick, 2025-05-09:
Thanks for the comment! To clarify, I don't believe it's a dependency at the moment (see [dependencies](https://github.com/isofit/isofit/blob/55161ed2887c6797a75ffdb77a86f71e90ce8abb/pyproject.toml#L32-L48)) - though as memory serves this is a relatively recent change.  

Is your suggestion to _add_ it as an optional dependency to one of the sets (or a new set), or maybe I'm missing something?

alexjohnson-kobold, 2025-05-10:
Ah my apologies! I was working with an old version without realizing it.

closed 2025-05-10 by alexjohnson-kobold