isofit/isofit #691: Hardcoded unit conversions when loading RTEs

evan-greenbrg, 2025-05-14:
Just talked with Niklas, but a few folks had run into issues when processing non-JPL data sources where the units of the input radiance files are not in the expected `uW/nm/cm2`. This can potentially cause issues in the inversion. We currently hard code expected units, which make debugging this unit-related issue difficult.

sRTMnet.py
```python
sol_irr = sol_irr / 10  # Convert to uW cm-2 sr-1 nm-1
```

modtran.py
```python
irr = tokens[18] * 1e6 * np.pi / tokens[8] / coszen  # uW/nm/cm2
```

We should discuss whether it makes sense to convert on the fly, or at least provide a warning message if the input radiance header either does not provide units, or provides units that don't match the assumed ISOFIT convention.

pgbrodrick, 2025-05-24:
I very much agree with the concept.  In practice, we'll have to think it through....it can be tricky to tell when we're only off by a factor of 10.  But minimally this feeds back into the comment about documenting the inputs more clearly.

pgbrodrick, 2025-06-30:
Resolved via #713

closed 2025-06-30 by pgbrodrick