isofit/isofit #730: Memory issue with Analytical Line

unbohn, 2025-07-13:
I'm still running into a memory issue on the latest v3.5.0 release. At full core utilization (128 out of 128) on TACC, I'm getting a `numpy.core._exceptions._ArrayMemoryError: Unable to allocate 1.38 MiB for an array with shape (425, 425) and data type float64` error at [line 361 of inverse_simple.py](https://github.com/isofit/isofit/blob/3b8b63b503eab33ff7ccd292e28af984018a682f/isofit/inversion/inverse_simple.py#L361). Seems like the prior covariance matrix is blowing up the memory here, even though it only requires a very small allocation. Reducing the number of cores to 64 solves the problem.

pgbrodrick, 2025-07-13:
How much memory do you have available on that 128 core node?

Can you also lay out your run parameterization?  I'm not seeing an issue on the systems I'm running on, but it's possible there's an exponential growth factor of some sort depending on the parameterization.

unbohn, 2025-09-30:
@jammont I reran the above call and the exact same memory issue still persist. I plotted the respective resources tracker [here](https://github.com/user-attachments/files/22626952/test.html). As you can see from the graphs, the aggregate memory starts to explode at some point (starting at time stamp 12:22:51). When comparing to the log file of that run (see attached below), this happens exactly when an incredible amount of duplicated statements appear: `Verifying path for ISOFIT data: /work/09147/bohn/ls6/python_modules/isofit_data/data`, ` Loaded ESD from file: /work/09147/bohn/ls6/python_modules/isofit_data/data/earth_sun_distance.txt`, and especially `SNR noise model found noise <= 0 - adjusting to slightly positive to avoid /0.`, which gets printed tens of thousands of times. The log file actually stops with this message 1:40 minutes later without entering the actual processing of the analytical line inference.

[ang20240815t160909_008.log](https://github.com/user-attachments/files/22626978/ang20240815t160909_008.log)

brentwilder, 2025-10-15:
Extrapolating a bit, but assuming @pgbrodrick array size in th e EMIT example was ~0.57mb (#744) , this would give a memory overhead of ~36GB for holding on to the max=500 cache. This is again worse case assuming the cache fills up for each worker.

unbohn, 2025-10-15:
@brentwilder Excellent observation! That could really be the issue here. This would explain why the memory error is not triggered during the superpixel inversion. As you mention, we use a [default hash_size of 50 set by the config](https://github.com/isofit/isofit/blob/dba8a44ba1f66885e8d62d99e854e846288a236f/isofit/configs/sections/implementation_config.py#L79), while the analytical_line uses a [hardcoded size of 500](https://github.com/isofit/isofit/blob/dba8a44ba1f66885e8d62d99e854e846288a236f/isofit/utils/analytical_line.py#L453). @pgbrodrick Any reason why we blew this up by a factor of 10 for the AOE solution? Anyway, reducing the number to the default value of 50 there might solve the problem.

brentwilder, 2025-10-20:
@pgbrodrick tested an EMIT image locally for the option 1 you are mentioning here.. Trimming hash from 500 to 50 per worker, cuts down performance (spectra/second/core) by ~6% for the analytical line solve, and uses ~ 20% less RAM in doing so.  Raw numbers were 
 
- 78.15 spectra/sec/core for hash=50 
 
- 73.74 spectra/sec/core for hash=500

closed 2025-10-30 by pgbrodrick